sgenoud/replicad #49: sweepSketch along a drawing not yet available?

raydeleu, 2022-07-10:
As mentioned in #47 I did not find an equivalent to `sweepSketch` in the drawing API. 

````javascript

// draw curve of hook with Drawing API
let hookCurveDraw = draw().cubicBezierCurveTo(tp1,cp1s,cp1e).cubicBezierCurveTo(tp2,cp2s,cp2e).cubicBezierCurveTo(tp3,cp3s,cp3e).cubicBezierCurveTo(tp4,cp4s,cp4e).done().sketchOnPlane("XY")

// draw curve of hook with Sketcher object
let hookCurveSketch = new Sketcher("XY",1).movePointerTo([0,0]).cubicBezierCurveTo(tp1,cp1s,cp1e).cubicBezierCurveTo(tp2,cp2s,cp2e).cubicBezierCurveTo(tp3,cp3s,cp3e).cubicBezierCurveTo(tp4,cp4s,cp4e).done()

// define cross section of hook
    let hookWidth =   3;
    let hookHeight =  2

// sweep along the sketch works, sweep along the drawing is not available yet? 
    let loftedHook = hookCurveSketch.sweepSketch((plane, origin) => 
    sketchRectangle(hookWidth, hookHeight, { plane, origin }) );

````

sgenoud, 2022-07-11:
As it is now you can just replace `hookCurveSketch` with `hookCurveDraw` - it works, doesn't it?

raydeleu, 2022-07-11:
You aren right, it works! Don't know what I did wrong earlier. I thought that in my earlier attempts the error message was that sweepSketch was not defined. This time I also got an error as I forgot to move the plane for the drawing in the Z-direction. In my code above you can see that the drawing is placed on the XY plane whereas the sketch was placed on the `XY,1` plane (The XY plane moved one unit up in the Z-direction). I close the issue then.

closed 2022-07-11 by raydeleu