sgenoud/replicad #54: init function from replicad_single.js expects 0 arguments

jirihon, 2022-07-27:
According to https://replicad.xyz/docs/use-as-a-library OpenCascadeInstance should be created by calling the init function exported from `replicad-opencascadejs/src/replicad_single.js` with single argument (defining locateFile function). However, the init function does not have any argument and TypeScript reports `Expected 0 arguments, but got 1.`. Could please help me to solve this?

paulftw, 2022-07-28:
I followed those instructions and it works fine in my project, but I'm not using typescript. 
Have you tried disabling the typescript checks? if it still doesn't work you may be calling a wrong function. If it does the problem is either in your TS config or project's type annotations. but the former is more likely.

It'd help if you provide a code snippet of what exactly you are calling and what arguments are passed.
without that this seems like a question and not yet an issue, so using github discussions would be more appropriate.

sgenoud, 2022-08-09:
I have updated the underlying library - can you confirm if it fixes your issue?

jirihon, 2022-08-10:
No, it does not fix the issue. The TypeScript definition of the init function is still the same - without any paramater. See https://github.com/sgenoud/replicad/blob/main/packages/replicad-opencascadejs/src/replicad_single.d.ts#L7165 It seems to be a bug in opencascadejs, so I should probably report there.