sgenoud/replicad #58: Exception in intersectBlueprints

paulftw, 2022-08-28:
`intersectBlueprints` throws `Cannot read properties of undefined (reading 'length')`. Input blueprints seem sane to me.

[online demo](https://scriptcad.com/paulftw/replicad-fuseAlgorithm-issue48)

```
import { BlueprintSketcher, fuse2D, intersectBlueprints } from 'replicad'

const triangles = [
  [
    [-15, -10],
    [4.22649730810374, 7.5],
    [-15, 30],
  ],
  [
    [-15, 30],
    [4.22649730810374, 7.5],
    [4.226497308103742, 30],
  ],
  [
    [7.690598923241496, 30],
    [7.690598923241495, 7.5],
    [15, 30],
  ],
  [
    [4.22649730810374, 7.5],
    [15, -10],
    [7.690598923241495, 7.5],
  ],
  [
    [-15, -10],
    [15, -10],
    [4.22649730810374, 7.5],
  ],
  [
    [7.690598923241495, 7.5],
    [15, -10],
    [15, 30],
  ],
]

const triangles2 = [
  [
    [9.999999999999996, 7.5],
    [-15, 30],
    [-15, -10],
  ],
  [
    [9.999999999999996, 7.5],
    [10, 30],
    [-15, 30],
  ],
  [
    [13.46410161513775, 7.5],
    [15, 30],
    [13.464101615137752, 30],
  ],
  [
    [15, -10],
    [13.46410161513775, 7.5],
    [9.999999999999996, 7.5],
  ],
  [
    [15, -10],
    [9.999999999999996, 7.5],
    [-15, -10],
  ],
  [
    [15, -10],
    [15, 30],
    [13.46410161513775, 7.5],
  ],
]

const trianglesToBlueprint = triangles => {
  const blueprints = triangles.map(pts => {
    const bps = new BlueprintSketcher().movePointerTo(pts[0])
    pts.slice(1).forEach(p => bps.lineTo(p))
    return bps.close()
  })

  return blueprints.slice(1).reduce((acc, el) => fuse2D(acc, el), blueprints[0])
}

export function main() {
  const a = trianglesToBlueprint(triangles)
  const b = trianglesToBlueprint(triangles2)

  // return [
  //   a.sketchOnPlane('XY').extrude(1).translate(0, 0, 20),
  //   b.sketchOnPlane('XY').extrude(1).translate(0, 0, 30),
  // ]

  return intersectBlueprints(a, b).extrude(1).translate(0, 0, 50)
}
```

sgenoud, 2022-09-13:
This pinpoints exactly why the `Drawing` abstraction is necessary. `intersectBlueprints` instersects two `Blueprint` object. But `fuse2D` returns a `Blueprints` object (with just one blueprint in it but you could fuse disjointed paths and have multiple blueprints).

I should tackle this: https://github.com/sgenoud/replicad/issues/46

Anyway, you can obtain what you want with:

```js
  return intersectBlueprints(a.blueprints[0], b.blueprints[0])
    .sketchOnPlane()
    .extrude(1);
```

sgenoud, 2022-09-13:
I have actually implemented #46 you can just replace `new BlueprintSketcher().movePointerTo(pts[0])` by `draw(pts[0])` and return `a.intersect(b)`.

sgenoud, 2023-02-27:
I think this has been fixed. You can have a look at [an example here](https://studio.replicad.xyz/workbench?code=UEsDBAoAAAAIABlPW1ZE68b20QEAAB0GAAAHAAAAY29kZS5qc5VUTW%252BcMBC98yusXGJLxLH5DIf2UKnn9pBDotUevN7ZBpUYZIwSKdr%252FXmO0rBdYkiKwbM%252Bb53lvAFmr1qAPtNfiDR3RN6ShqUop9kEgXcjoUqg%252FFbQ2tgmQe%252Bx4x9MQ3XG2DYd1QqMoS4o8Zg%252BcxXkSopymp6ADxwPWDZv5%252Fmcc02C0yJjTrGBp8VBEcZTwpMh8%252FmkwvTjgSo2rRU1MWDlgJttL%252B7KVa0rnYiYnnLVtZ72NLptb0OLiyq52c6rGr3CdhbMZyYL7PKZJlnDGM57yOM%252BvtmwJu%252FyKTK1ZPWFFwwrnF%252FxbMux609brXGzpY%252F2j6qDRpTK2ud5X%252FB192JwBvTtBWh9DX0WDGzNiR3TTwxS8ub9Fj9iwLXEAO6et%252FW8A5oQeav1TyBfc9AQ2iValgscaN2QAazCdVi4iq7oF3G8fSRCcQ2NdZ1YN%252B87OsJAyRFCRntzO6aGzDHYdelmurmMQwHtTa4MOnZKmrBV6FaXCxDNA%252BLo9y%252FC4Sc5mfYqNnIb7%252B5OMzbBCSND2Lxj58kv9roQCfPv0fEsovBvd7Z04o4VqK2EA28%252FC3hEj4Sl597%252FJ8Zi89TwV1NYKugVp8I5MOG%252Benm9ClDKfNzj%252BA1BLAQIUAAoAAAAIABlPW1ZE68b20QEAAB0GAAAHAAAAAAAAAAAAAAAAAAAAAABjb2RlLmpzUEsFBgAAAAABAAEANQAAAPYBAAAAAA%253D%253D)

closed 2023-02-27 by sgenoud