sgenoud/replicad #61: Potential bug in cast() or downcast() of shapes.ts

nine-fox, 2022-10-09:
Hi,

I just found, the following cast() or downcast() code in shapes.ts will emit an error sometimes - not sure why, it happens accidentally.

Error: Could not find a wrapper for this shape type

Not sure why, I believe it is a comparation issue for java object instances. Since the value of enum is just the XXX.value, so I would suggest to fix like this(add .value call):

export function cast(shape: TopoDS_Shape): AnyShape {
  const oc = getOC();
  const ta = oc.TopAbs_ShapeEnum;

  const CAST_MAP = new Map([
    [ta.TopAbs_VERTEX.value, Vertex],
    [ta.TopAbs_EDGE.value, Edge],
    [ta.TopAbs_WIRE.value, Wire],
    [ta.TopAbs_FACE.value, Face],
    [ta.TopAbs_SHELL.value, Shell],
    [ta.TopAbs_SOLID.value, Solid],
    [ta.TopAbs_COMPSOLID.value, CompSolid],
    [ta.TopAbs_COMPOUND.value, Compound],
  ]);

  const Klass = CAST_MAP.get(shapeType(shape).value);

  if (!Klass) throw new Error("Could not find a wrapper for this shape type");
  return new Klass(downcast(shape));
}

sgenoud, 2022-10-11:
I would be curious to have an example of when it breaks so I can debug it. One guess would be that you use two different versions of replicad / opencascadejs - is it something you sometimes do?

nine-fox, 2022-10-13:
> I would be curious to have an example of when it breaks so I can debug it. One guess would be that you use two different versions of replicad / opencascadejs - is it something you sometimes do?

Hi, thanks for your asking.

I am using the same version of replicad / opencascadejs 0.13.2(latest for now) and 0.13.0.

Since the calculation for CAD generation is time-cost, I am using the multi workers. Different workers have seperate wasm loading procedure. I guess it is the reason for this problem.

My question is, would it be better to test equal with raw value, rather than the java object? In my opinion, there's no harm to do this.