sgenoud/replicad #63: Can't save from visualizer?

lf94, 2022-12-11:
For some reason, currently trying to save from replicad.xyz just doesn't cause anything to download. Devtools console shows some sort of error:

[image]

Most likely a very basic issue but an issue nonetheless :)

My rcad file may be a culprit but I'm doubtful:

```js
/home/lee/Code/lee/Kindle-Scribe-Case/main.rcad:1.1,58.1
const extrusion = {
  width: 0.6,
};

const base = {
  dim: [196, 230],
  height: 5.8,
};

const cover = {
  dim: [base.dim[0] + extrusion.width * 2, base.dim[1] + extrusion.width * 2]
};

const foot = {
  offsets:  [
    [base.dim[0] /  2 + -18, base.dim[1] / -2 +  12],
    [base.dim[0] / -2 +  18, base.dim[1] / -2 +  12],
    [base.dim[0] /  2 + -18, base.dim[1] /  2 + -12],
    [base.dim[0] / -2 +  18, base.dim[1] /  2 + -12]
  ],
  diameter: 3.8,
  height: 7.0,
};

const main = ({ drawRoundedRectangle, drawCircle, Drawing }, {}) => {
  const drawingBase = drawRoundedRectangle(...base.dim)
  const drawingCover = drawRoundedRectangle(...cover.dim)
  const drawingFoot = drawCircle(foot.diameter/2.0);

  const shapeCover = new Drawing()
    .fuse(drawingCover)
    .cut(drawingFoot.translate(foot.offsets[0]))
    .cut(drawingFoot.translate(foot.offsets[1]))
    .cut(drawingFoot.translate(foot.offsets[2]))
    .cut(drawingFoot.translate(foot.offsets[3]))
    .sketchOnPlane("XY")
    .extrude(base.height + foot.height / 2.0)
    .fillet(6.5, e => e.inDirection("Z"))
    .fillet(2.0, e => e
      .inBox([cover.dim[0] / -2, cover.dim[1] / -2, 0], [cover.dim[0] / 2, cover.dim[1] / 2, base.height / 2])
      .not(f => f.ofCurveType("ELLIPSE"))
    )
    .fillet(1.5, e => e.inBox([cover.dim[0] / -2, cover.dim[1] / -2, base.height], [cover.dim[0] / 2, cover.dim[1] / 2, base.height * 2]));

  const shapeBase = drawingBase
    .sketchOnPlane("XY")
    .extrude(base.height)
    .fillet(6.5, e => e.inDirection("Z"))
    .fillet(2.0, e => e
      .inBox([base.dim[0] / -2, base.dim[1] / -2, 0], [base.dim[0] / 2, base.dim[1] / 2, base.height / 2])
      .not(f => f.ofCurveType("ELLIPSE"))
    )
    .fillet(1.5, e => e.inBox([cover.dim[0] / -2, cover.dim[1] / -2, base.height / 2], [cover.dim[0] / 2, cover.dim[1] / 2, base.height]));

  const result = shapeCover.cut(shapeBase.translate([0, 0, foot.height / 2.0]));

  return result; // tried with [result] also
}
```

lf94, 2022-12-11:
Looking deeper with a debugger, the real error is "Security error". Going to try disabling security in Chromium to see if it'll work then.

lf94, 2022-12-11:
`[145337:145360:1211/010134.970498:ERROR:select_file_dialog_linux_portal.cc(718)] Portal returned error: org.freedesktop.portal.Error.InvalidArgument: invalid filter: name is empty` from when I start chromium from a terminal; seems maybe there was a slight change here?

sgenoud, 2022-12-29:
I have made the change - can you confirm that your issue is fixed?

lf94, 2023-01-06:
Nice, works again! JSON download is still broken but I only care about STL :D

sgenoud, 2023-01-06:
It should be fixed for JSON as well!

closed 2023-01-06 by sgenoud