sgenoud/replicad #67: How to build RepliCAD?

lf94, 2023-01-06:
In the event of broken code or Internet outage, it'd be nice to be able to build a local copy of RepliCAD. The latest issue has caused me to start to look at other options (such as using OpenCASCADE in StandardML). I would like to not have to do this... It's been a cycle of jumping between code CADs and RepliCAD hits some nice points!

sgenoud, 2023-01-06:
Have you had a look at the [sample app](https://sample-app.replicad.xyz/)? You can easily modify it and run it locally (for when the broken code issue).

Concerning the internet outage you should be covered. The visualiser is a PWA and should work even if the internet does not work anymore (I'd be curious if it does not work).

lf94, 2023-01-06:
I think I did but was more so using it to try to get the replicad project to build - I guess I should have used the sample project instead entirely :sweat_smile:

closed 2023-01-06 by sgenoud