sgenoud/replicad #72: Interest in using SolveSpaces solver?

vespakoen, 2023-02-20:
I just uploaded JS / WASM bindings for SolveSpace in my PR.

This might be fun to play with over here?

https://github.com/solvespace/solvespace/pull/1343#issuecomment-1437618758

The API will probably change a bit before it gets merged, but i'd be interested to hear your thought ;)

nine-fox, 2023-03-23:
> I just uploaded JS / WASM bindings for SolveSpace in my PR.
> 
> This might be fun to play with over here?
> 
> [solvespace/solvespace#1343 (comment)](https://github.com/solvespace/solvespace/pull/1343#issuecomment-1437618758)
> 
> The API will probably change a bit before it gets merged, but i'd be interested to hear your thought ;)

Hi @vespakoen I am quite intertested in solvespace wasm, since opencascade is too heavy for me. 

However, I did some investigation on solvespace compilation on wasm, didn't find any code for it.

Would you please let me know where to find related resources? Thanks

vespakoen, 2023-03-23:
There is an experimental WASM version hosted on NPM (https://www.npmjs.com/package/slvs?activeTab=versions), try version [3.1.0-dev.22](https://www.npmjs.com/package/slvs/v/3.1.0-dev.22), it might work for you.
The code is in the linked pull request, see: https://github.com/solvespace/solvespace/pull/1343/files#diff-19f291f169fa52288f5bb45776872abd63c7f575975d862eefa3243ca43acece for the bindings for example.

NodeGuy, 2023-04-02:
Excellent!  So how would we use this in Replicad?  Pipe its output to [the Solid.translate method](https://replicad.xyz/docs/api/classes/Solid#translate)?

hernanponcetta, 2023-04-06:
@nine-fox take a look at [Three.cad](https://github.com/twpride/three.cad). That project is using SolveSpace.

nine-fox, 2023-04-06:
> @nine-fox take a look at [Three.cad](https://github.com/twpride/three.cad). That project is using SolveSpace.

hi @hernanponcetta Thanks a lot for your information! That helps. After investigation, I just found three.cad can not export STEP yet. Since support STEP is a MUST option in my case, do you know exporting STEP can be supported by solvespace wasm? Thanks :)