sgenoud/replicad #74: Replicad Share Application doesn't put all objects into same coordinates.

joeynelson, 2023-02-23:
I'm new to the library and no very javascript savvy, so this might be a user error.

In the share app it seems to space all the models out.
https://studio.replicad.xyz/share/https%3A%2F%2Fgist.githubusercontent.com%2Fjoeynelson%2Ff4b6915c720cab12c85c966021348257%2Fraw%2Ff44a809bc21fde4298334994bf285c851bf9a17b%2Fgistfile1.txt

In workbench it keeps them in a common coordinate system.
https://studio.replicad.xyz/workbench?code=UEsDBAoAAAAIAOurV1ZbZWGZNwIAAH8EAAAHAAAAY29kZS5qc3VTTY%252FaMBC98ytGnOxiQpJd6AKlF9o9VeqqvbRdocrrDGCtcSLHEYsQ%252F72280GotIckE3vmzXvzIXJdWjjDgb%252Fi%252BqSkztAwyAw%252FrqURCmv7BwrL9a79vcAKDBZKCp4tBwMRMDLc8krZJ274oXQO5wFAKbjWaNZ5pe0C7pk%252FckBZvt0uIH3w%252FyrffZH8gBaNO4oHFwfYIBa54i62kOjxSLnnhSMgPBgDwzNZlQzyggtpT3%252BlXiUMpJZWcrWKKaw%252BBwo11Asv0WEEiMgarkvFLZK4xaHLzjXodL53sxgmdTZPqb0WLZ%252FnzTWm4MqfDA1mQwbDnUHU3nhRFfrvCZXKj9468B1qy70pTlwPA8g2N0AUWpAOJV66z6cmMcjRiAYd0GSOiqrck3NQsvCyIqFyjYRGJrdek4QPtYZRUwzqa6Zys3Asn%252BWmK9miV7tLKMDFPQZtZXSTbBnaUWs8vj0prn1lSFfcxtnPBKGBJEDkhghJOh1%252FjKbsLumO99%252F8%252BXga3zo2npSNe75OUolXxPIVrdh%252F14EAGf76Peyu8M2aKnOdjK7R1%252F56Csl9d9FUKHmIaa1sMNhWWliZa7cAUpMz%252FAyp%252FA48coGPzT58zXadXZOptwMu7NyfcdaNN%252BsP9qVuYTcuTYifot6Mk5tNI0k0S1kazaf0P%252F20E53M6VXsHzJO5pPUdfs9RgmwOPS5adtzf%252BlJj%252B8kZUka32LPYrqJHH3BLemTbsaCUPZe2jhK3YvSNrpV73vwD1BLAQIUAAoAAAAIAOurV1ZbZWGZNwIAAH8EAAAHAAAAAAAAAAAAAAAAAAAAAABjb2RlLmpzUEsFBgAAAAABAAEANQAAAFwCAAAAAA%253D%253D

sgenoud, 2023-02-24:
This is expected behaviour - but I might want to change it. 

The idea with the share functionality is to show the models - it automatically puts them on a grid and ignores its coordinates. 

You could fuse all your pieces together (with `.fuse` or `makeCompound`)  so that they keep their positions (but you would loose the different colors). 

Generally, replicad does not have a good assembly story yet. 

I should add an option to the sharing widget to keep the coordinates. What is your use case (so I can understand the best way of doing this).

joeynelson, 2023-02-24:
That makes sense.  What I'm trying to do is allow people to put parameterize descriptions of multiple 3D laser scanners for a system, and then be able to view a 3D representation of the system.  Being able to have multiple objects with different colors and opacities is important for scanning devices, fields of view, target object, and other scene objects.

My use case is a bit of an edge case for Replicad, and there may be more appropriate tools for this.  However Replicad was the easiest to make a proof of concept that I immediately share on the web.   Another option would be to build from the web app.  Currently it looks like it only handles a single object, but it might not be that hard to add multiple object support.  Most likely we will end up wanting a more customized interface if we continue with this project.

sgenoud, 2023-02-24:
I should have more time Monday to look at it, but it should be fairly easy to do. 

From how I think of replicad, your use case is exactly the kind of things I have been building it for. And in case you continue this project and need some consulting, don’t hesitate to contact me!

joeynelson, 2023-02-24:
Thanks.  Having the simple configuration interface of the share app would be nice.  If we do go ahead I'll likely reach out.

sgenoud, 2023-02-27:
The modifications have been done. Have a look [here for instance](https://studio.replicad.xyz/share/https%3A%2F%2Fgist.githubusercontent.com%2Fjoeynelson%2Ff4b6915c720cab12c85c966021348257%2Fraw%2Ff44a809bc21fde4298334994bf285c851bf9a17b%2Fgistfile1.txt?disable-auto-position=true&disable-damping=true). This should work once you reload the version of the app.

The share screen as a couple of additional options as well depending on how you want to present your model.

[image]

joeynelson, 2023-03-01:
Awesome!

closed 2023-03-01 by joeynelson