sgenoud/replicad #89: extrusion of sketches goes in the wrong direction

tve, 2023-07-23:
In the following example I'm extruding a sketch along the Y axis but a positive extrusion value results in extruding towards -Y. A subsequent translateY of the object has the direction of the Y axis correct. [Example](https://studio.replicad.xyz/workbench?code=UEsDBAoAAAAIANeU91ZNp801vgAAAGUBAAAHAAAAY29kZS5qc62PMWoDMRBF%252Bz3FoEqCRewupAnkBoG4cOHEuBhrx1hEHi2jUWIwvntkV4Y0KdK9D%252F8%252FZui8ZFGY6YA1KRwqB42Z4YSR7eLg0gGEzEVBJY7wAkJLigFnPwt%252BW%252Be%252FXiOTHQfnU4N1tttx6Ied8yHlQq1QPknD8Y1XCVvRbD6M83RWqfN99uif%252Ftv%252FdNMLaRWGbUOAC5QjLvR8f6cHxlNjcwumb4ekLC3uUyUD1%252F7XYvIqyCWh0rudBvcHwa67dj9QSwECFAAKAAAACADXlPdWTafNNb4AAABlAQAABwAAAAAAAAAAAAAAAAAAAAAAY29kZS5qc1BLBQYAAAAAAQABADUAAADjAAAAAAA%253D):

```
export default function main(p) {
  const tri1 = replicad.draw().vLine(10).lineTo([10,0]).close().sketchOnPlane("XZ").extrude(10)
  const tri2 = replicad.draw().vLine(10).lineTo([10,0]).close().sketchOnPlane("XZ").extrude(5)
  return [
    { shape: tri1, name: "tri1", color: "blue" },
    { shape: tri2.translateY(20), name: "tri1", color: "blue" },
  ]
}
```

[image]

sgenoud, 2023-07-30:
This is due to the (mathematical definition of the XZ plane). If you do the cross product of X (1, 0, 0) and Z (0, 0, 1) you obtain -Y (0, -1, 0). When sketching in a plane, you will, by default extrude in the direction of the normal of the plane, -Y in this case.

That said, I completely understand the confusion, and I am not sure if that is the best choice (i.e. to be mathematically consistent instead of intuitive). There might be some edge case that pop when we start ignoring mathematical consistency.

The ZX plane is the one you where expected to get...

tve, 2023-07-31:
Ah, right hand rule... I forgot... I'll get used to using ZX instead of XZ.
I fear that if you make it "intuitive" via some special rule things will either be wrong or really confusing when someone uses a pivoted plane... E.g. what if I take XZ and pivot it 120 degrees around the Z axis, now what's + vs. -?

closed 2023-07-31 by tve