sgenoud/replicad #93: Cutting a fused shape shows only edges

bramvandooren, 2023-07-31:
When trying the code below to cutout a fused circle with text it only show the edges of the fused shape instead of it being cut.

```
const main = (r) => {
       function center(drawing) {
            const boundingBox = drawing.boundingBox;
            drawing = drawing.translate(-boundingBox.center[0], -boundingBox.center[1]);

            return drawing;
        }

        let plate = drawRectangle(5, 5).cut(
            drawCircle(0.5)
                .fuse(center(drawText('test', { fontSize: 2 })).translate(0, 0.8).rotate(90))
        );

        return plate.sketchOnPlane('XY').extrude(0.2);
};
```
[image]

raydeleu, 2023-10-19:
A simple workaround is not to expect the drawing to understand which parts should be kept and which should be deleted.

``` javascript
const {draw, drawRectangle, drawCircle, drawText} = replicad

const main = (r) => {
       function center(drawing) {
            const boundingBox = drawing.boundingBox;
            drawing = drawing.translate(-boundingBox.center[0], -boundingBox.center[1]);

            return drawing;
        }

        let plate = drawRectangle(5, 5).sketchOnPlane("XY").extrude(0.2)
        let cutter = drawCircle(0.5).fuse(center(drawText('test', { fontSize: 2 })).translate(0, 0.8).rotate(90))
        cutter = cutter.sketchOnPlane("XY").extrude(1.0);
        plate = plate.cut(cutter)

        return plate;
};
```

bramvandooren, 2023-12-19:
Thank you for your response. I've explored the option of cutting after extrusion for this particular scenario, and while it may not be the optimal solution for my project, it serves as a viable workaround for the time being. Your assistance is greatly appreciated.

sgenoud, 2023-12-20:
I confirm the whole issue - and I have been working on a better solution for 2D booleans (which can be faster) - but nothing that I would recommend yet unfortunately.