sgenoud/replicad #101: Add Support for Exporting Reference Geometry

andrewkrippner, 2023-08-27:
It would be helpful to export axes and planes in a STEP file to mark locations and mounting points etc. Happy to sponsor if possible.

sgenoud, 2023-08-27:
I am quite busy until the end of year - but I'll see what I can do. I guess you are talking about a standard CAD feature I am not aware of (I came into it from the hobbyist side). Can you point me to an implementation of this feature (i.e. how it is called in Fusion360 or FreeCAD for instance).

andrewkrippner, 2023-08-27:
I actually was able to make reference planes in your demo here: https://github.com/sgenoud/replicad/blob/a1f85c09babb215e7a0df0b72578e7bdd498a76d/packages/replicad-docs/docs/tutorial-overview/planes-and-sketches.md?plain=1#L34

This problem would be solved if we could export all of these planes in a single step file, rather than three separate ones as this does. Is that possible?

sgenoud, 2023-08-28:
It is possible - but it will be in the same colour. Something[ like this.](https://studio.replicad.xyz/workbench?code=UEsDBAoAAAAIALWZHFfc1tg%252FnAAAACIBAAAHAAAAY29kZS5qc32OsQ6CQAxAd7%252BiYbpLiMHBieDiB2iYBOPQ3FUlQCHHEQfDv1tPBzVq0u29vhZMx4OHK1iHl7wb2ZLNyXjkU0MxtFjTumv7O4AJMnDUN5VBm8IMZB7bLVYsTGnIVpIS8gROSgK%252BtdUiSWJYJjqUxPSj47d7ah8IhMp8qMmb84a3DTKpaFdEen5EQ4PS8T%252BvfPF%252Ba0X5mTuEx6YUblBLAQIUAAoAAAAIALWZHFfc1tg%252FnAAAACIBAAAHAAAAAAAAAAAAAAAAAAAAAABjb2RlLmpzUEsFBgAAAAABAAEANQAAAMEAAAAAAA%253D%253D)

andrewkrippner, 2023-08-28:
Thats actually perfect, that's all I was looking for thank you!

closed 2023-08-28 by andrewkrippner