sgenoud/replicad #102: Section view broken for some models, hard to fix

Billiam, 2023-08-27:
A model I was working can't use section view. The preview window shows the message "Something went wrong.", and the console reports this:

```
index-9310d50d.js:4546 TypeError: (void 0) is not a constructor
    at index-9310d50d.js:3852:11254
    at $h (index-9310d50d.js:3812:20924)
    at ku (index-9310d50d.js:3812:40498)
    at rO (index-9310d50d.js:3812:34730)
    at Ns (index-9310d50d.js:3810:4702)
    at C9 (index-9310d50d.js:3812:39309)
    at Ru (index-9310d50d.js:3812:38361)
    at nO (index-9310d50d.js:3812:33843)
    at T (index-9310d50d.js:3798:1561)
    at MessagePort.F (index-9310d50d.js:3798:1927) 
Object
componentStack
: 
"\n    at https://studio.replicad.xyz/assets/index-9310d50d.js:3816:34807\n    at aj (https://studio.replicad.xyz/assets/index-9310d50d.js:3816:32530)\n    at m (https://studio.replicad.xyz/assets/index-9310d50d.js:3611:6702)\n    at Suspense\n    at kP (https://studio.replicad.xyz/assets/index-9310d50d.js:3820:25)\n    at https://studio.replicad.xyz/assets/index-9310d50d.js:3674:2792\n    at https://studio.replicad.xyz/assets/index-9310d50d.js:3674:2792\n    at c1e (https://studio.replicad.xyz/assets/index-9310d50d.js:4546:9029)\n    at div\n    at m (https://studio.replicad.xyz/assets/index-9310d50d.js:3611:6702)\n    at div\n    at m (https://studio.replicad.xyz/assets/index-9310d50d.js:3611:6702)\n    at bz (https://studio.replicad.xyz/assets/index-9310d50d.js:4617:17)\n    at div\n    at div\n    at kW (https://studio.replicad.xyz/assets/index-9310d50d.js:4546:3634)\n    at https://studio.replicad.xyz/assets/index-9310d50d.js:3674:2792\n    at cV\n    at div\n    at m (https://studio.replicad.xyz/assets/index-9310d50d.js:3611:6702)\n    at Awe\n    at Suspense\n    at e (https://studio.replicad.xyz/assets/index-9310d50d.js:3600:8548)\n    at e (https://studio.replicad.xyz/assets/index-9310d50d.js:3600:9517)\n    at Axe\n    at e (https://studio.replicad.xyz/assets/index-9310d50d.js:3600:5723)\n    at e (https://studio.replicad.xyz/assets/index-9310d50d.js:3600:10581)"
[[Prototype]]
: 
Object
```

When this occurs, it doesn't seem like there's a (UI) path to fixing the issue, short of reloading.

## Steps to reproduce:

1. Visit model url: https://studio.replicad.xyz/workbench?code=UEsDBAoAAAAIAASYG1fIglyYsgIAAGYHAAAHAAAAY29kZS5qc3VUXWvbMBR9968QeShyo7hJ28AIpGNtN%252FYy6MNgH6UM1b5pRBW5k%252BWlpeS%252F7%252BrDsp04UNLofpxz79FRkrxUlSHvpNB8y8iGP8M1r%252BC6fPWHmzcpVAGa7MiSaHiRIudFkviuAla8luaOa76pMP%252BeEFIIvgEDekFmswuGAbMW%252BbOCqlqQD9mljeRlrcyCzJkr%252FycQHpNG15DsGuhcAqKqHBB2ml2E6JZL%252Bb3Bw0wTX8my1EOJAN9NTbN5Q7LhQmGE4hzaDmPnbzdg7hTH90c3Owt1fnQ87JKULK9cv0d%252BLF9%252FiMKsEf0bN%252Bus%252BqsNdd9eyi2lDQUZt4umZ%252BeMnKdkQmJdLDvFqWc2m%252BLXc%252BzqC4GxLvNXEE9rg9S9%252FstYEga%252FsatglVsJaWdJrBAKu0Sp0Qa38OLWoFGI3tAI7fvHhDaCkI%252BHup%252F2WRdkmlo6CW5ixO84jzp5Gwm92AcTDWjAmj4vAJ7SuFFem7I2gacxtSc6ehuDxD56P2XE%252Fj20x5k9Ws7MIEIluQGK4QndR3AXvefY8ZEp3M3NnVReJvzMcBfq93EJsepIf3LSF%252FoKhQ6%252B9gAKtuST1vyNduvSbCWkpKqWMs00FHUOlD4y8gckIyKa27%252BAjoeGLq57eWir3jWxWLHvENbtDQ5GT85CuKtqrJy2TZN9lc%252FsM9knccFBI%252BN3gWIPu3wAKW2p44jhvwZTa0Ues1VdQaOyT%252B6Y3S51vxgHj%252B1z8QRf%252FI%252FtksDyCjKhoqj3k0ZStwOabWjjiU109cOfjcainX5XNihYr33s2t0z6vrPwTm%252FgKFoZbDusMPeCg25EaWio9%252Bj1G%252Bc5Wu%252BWeFTQ%252BSw053kCujo568Ra9%252FRsXKuCnofhB3oj8PG%252BzhU0yUeUvc0w9SH6MdnixQPgcOVqtLQQypLYldJWhegXrvkP1BLAQIUAAoAAAAIAASYG1fIglyYsgIAAGYHAAAHAAAAAAAAAAAAAAAAAAAAAABjb2RlLmpzUEsFBgAAAAABAAEANQAAANcCAAAAAA%253D%253D
2. Click section view button in toolbar

sgenoud, 2023-08-27:
Thanks, it seems a problem with every model - I will have a look!

Niels-IO, 2023-11-16:
This can be closed, it works now in the deployed version 🙂

closed 2023-11-16 by sgenoud