sgenoud/replicad #111: Import STEP file in my scene

maelp, 2023-10-16:
Hi,

I'd like to use replicad to build an object, but one of the elements of this object should be imported from an existing mesh (as .glb or .step file), is that possible?

maelp, 2023-10-16:
Just saw this so I guess I can close the issue 😅
https://github.com/sgenoud/replicad/blob/main/packages/replicad/src/importers.ts#L13

closed 2023-10-16 by maelp

raydeleu, 2023-10-17:
Can you show an example how you use this function? I struggle to make the
workbench read a file and display the shape. I tried something like the
code below but it doesn't work.

const {importSTEP} = replicad;

function main()
{

async function readFile(file) {
try {
const contents = await file.text();
const blob = new Blob([contents], { type: 'text/plain' });
return blob
}
catch (err) {
console.error('Error reading file:', err);
}
}

let bigblob = readFile("/Users/test/Downloads/holder.step");
let shape = importSTEP(bigblob);

return shape
}