sgenoud/replicad #122: Some typing issues

maelp, 2023-12-18:
When trying to compile the equivalent of cad.js in typescript I get this error:

```
error TS2339: Property 'shell' does not exist on type 'Shell | Solid | CompSolid | Compound | Vertex | Edge | Wire | Face'.
  Property 'shell' does not exist on type 'Vertex'.
```

maelp, 2023-12-18:
Similarly, when trying to compile 

```
  const OC = await opencascade({
    locateFile: () => opencascadeWasm,
  });
```

there's an error

```
error TS2554: Expected 0 arguments, but got 1.
```

(and indeed the .d.ts shows `declare function init(): Promise<OpenCascadeInstance>;`)

sgenoud, 2023-12-18:
What is the context of the first one?

For the second one I looked into it at some point - if I remember well, it comes from the [upstream project ](https://ocjs.org/)

maelp, 2023-12-18:
I just tried to compile this in Typescript:

https://github.com/sgenoud/replicad/blob/main/packages/replicad-app-example/src/cad.js#L4

and that was the error shown