sgenoud/replicad #126: BUG: makeBaseBox

maelp, 2023-12-23:
Isn't it a bit weird that the baseBox should be centered on the origin in x / y dimensions, but will not be centered in z dimension?

https://github.com/sgenoud/replicad/blob/c80d81918617dd10d0f004f750e180396390f30d/packages/replicad/src/shortcuts.ts#L10

perhaps it should be this?

```
const makeBaseBox = (
  xLength: number,
  yLength: number,
  zLength: number
) => {
  return new Sketcher()
    .movePointerTo([-xLength / 2, yLength / 2])
    .hLine(xLength)
    .vLine(-yLength)
    .hLine(-xLength)
    .close()
    .extrude(zLength).translateZ(-zLength/2);
};
```

maelp, 2023-12-23:
ah it seems that "extrude" extrudes "symetrically around the center", eg a vertical extrusion of length L will extrude L/2 in the vertical direction and L/2 in the opposite direction? Perhaps it should be documented

raydeleu, 2023-12-23:
I have explained the `extrude` method here https://github.com/raydeleu/ReplicadManual/wiki/5.-Create-solid-shapes#521-extrude. I don't think that you are correct that the extrude is always symmetrical. 

You are correct that it is a bit odd that the box is not centered on the origin, but I assume that is why the programmer calls it `makeBaseBox`. It is a box with the base flat on the XY-plane. So it is probably not a bug but a feature :-)

maelp, 2023-12-23:
Thanks, then I'm closing the issue (although I think the makeBaseBox behavior should probably be explained in some doc)

closed 2023-12-23 by maelp