sgenoud/replicad #127: Missing UV mapping on the geometry

maelp, 2023-12-24:
When I do this

```
const geometries = syncGeometries([replicadMesh], []);
const geometry = geometries[0].faces;
```

and use the `geometry: BufferGeometry`, it seems to miss the `uv` field in `geometry.attributes`

Do you know if there's an easy way to obtain those?

maelp, 2023-12-24:
Okay I fixed it using this

```
const updateUVMap = (geometry: THREE.BufferGeometry) => {
geometry.computeBoundingBox();
const max = geometry.boundingBox.max;
const min = geometry.boundingBox.min;
const offset = new THREE.Vector2(0 - min.x, 0 - min.y);
const range = new THREE.Vector2(max.x - min.x, max.y - min.y);
const positions = geometry.attributes.position.array;
const uvs: number[] = [];

for (let i = 0; i < positions.length; i += 3) {
    const x = positions[i];
    const y = positions[i + 1];
    uvs.push((x + offset.x) / range.x, (y + offset.y) / range.y);
}

  geometry.setAttribute('uv', new THREE.BufferAttribute(new Float32Array(uvs), 2));
geometry.uvsNeedUpdate = true;
};
```

sgenoud, 2024-01-03:
I am not 100% sure I understand how UV and replicad should interact (because I don't really understand them). Is there something you think you be done to improve the situation?

maelp, 2024-01-03:
Not exactly sure how it should work, but the snippet above fixed my issue

sgenoud, 2024-01-03:
It looks like this issue is related: https://github.com/sgenoud/replicad/issues/62 (in case you feel like digging deeper).

closed 2024-01-09 by maelp