sgenoud/replicad #137: Set a name for parts / the default scale in the output file

maelp, 2024-01-09:
I think that OpenCascade allows you to set the scale of the file (eg. meters or milimeters), when exporting to STEP / STL, is there a way to set that from replicad?

Also wondering if there's a way to go through all the parts and give them colors / names ?

sgenoud, 2024-01-09:
> I think that OpenCascade allows you to set the scale of the file (eg. meters or milimeters), when exporting to STEP / STL, is there a way to set that from replicad?

No, I have not exposed them now (but I don't see any reason on the top of my head not to).

> Also wondering if there's a way to go through all the parts and give them colors / names ?

You can give the parts within the workbench some names and colors, but this is not done at the opencascade layer (and will not be shown in the STEP file). Not sure how I would expose that though.

> BTW what's the proper way to access the Opencascade.js underlying implementation from Replicad, and the meshes? I'm not sure exactly how the wrapper is built and how I can call Opencascade.js functions if needed?

The [`getOC`](https://replicad.xyz/docs/api/#getoc) function returns the opencascade object that you can use directly.

maelp, 2024-03-19:
Very cool!! Thanks a lot!