sgenoud/replicad #142: How to get the volume and surface area of a shape?

Niels-IO, 2024-02-09:
Hi @sgenoud,

Is there any way to get the volume and surface area of a solid shape with Replicad?

Thanks a lot!

kriho, 2024-04-02:
I would also like to know if this is possible!

sgenoud, 2024-04-02:
You could use opencascade directly to do this:

```js
const {  getOC} = replicad;

export function measureVolume(shape) {
  const oc = getOC();
  const properties = new oc.GProp_GProps_1();
  oc.BRepGProp.VolumeProperties_1(
    shape.wrapped,
    properties,
    false,
    false,
  );

  const volume = properties.Mass();
  properties.delete();
  return volume;
}
```

Not very user friendly. So I added a set of measure helper to do that `measureVolume`, `measureArea` and `measureLength`. Make sure to have the latest version of the studio and use it on shapes!

kriho, 2024-04-04:
Thanks a lot for implementing this directly! I tried using measureVolume() but the function never returned (maybe I was too impatient and it just takes minutes).

closed 2024-07-12 by sgenoud