sgenoud/replicad #159: Intersecting text taking a while

r-bt, 2024-06-29:
Hi,

Firstly, thanks for the awesome library.

I am trying to extrude and intersect two pieces of text:

```javascript
const { drawText } = replicad;

const main = () => {
  
  const john = drawText("JOHN", {
    startX: 0,
    startY: 0,
  })
    .sketchOnPlane("XZ")
    .extrude(50);

  const mary = drawText("MARY", {
    startX: 50,
    startY: 0,
  })
    .sketchOnPlane("YZ")
    .extrude(50);

  const combined = john.intersect(mary);

  return [
    combined
  ]

};
```

In the workbench, however, it takes a while to complete, around ~2-5 minutes which is longer than I expected such an operation to take.

sgenoud, 2024-07-05:
Unfortunately, boolean operations with the base library I use (open cascade) are relatively slow.
That said, your code run on my machine (4 years old intel macbook pro) in about 10 seconds. Do you use a particular browser / machine?

r-bt, 2024-07-06:
Interesting, that's unfortunate as the library is great otherwise. I tried doing the same thing in JSCad but it doesn't have as nice support for extruding text or directly rendering in the browser. 

I'm using a Macbook Pro with the Apple M1 Pro, I tried on both Firefox and Chrome and they were both equally slow (ranging from 1 to 5 minutes)