sgenoud/replicad #163: Error when installing packages for replicad-app-example project.

mmiscool, 2024-09-04:
Hello, 

I am running Ubuntu 22.04.4 LTS with Node v20.17.0 and npm 10.8.2

```
git clone git@github.com:sgenoud/replicad.git
cp -r replicad/packages/replicad-app-example .
cd replicad-app-example
npm install
npm start
```
Running the instructions from the replicad-app-example readme results in a dependency error related to the threejs packages. 
```
git clone git@github.com:sgenoud/replicad.git
cp -r replicad/packages/replicad-app-example .
cd replicad-app-example
npm install
npm start
```

Error:
```
replicad-app-example$ npm install
npm error code ERESOLVE
npm error ERESOLVE unable to resolve dependency tree
npm error
npm error While resolving: replicad-app-example@0.17.1
npm error Found: three@0.148.0
npm error node_modules/three
npm error   three@"0.148.0" from the root project
npm error
npm error Could not resolve dependency:
npm error peer three@">=0.155.0" from replicad-threejs-helper@0.17.0
npm error node_modules/replicad-threejs-helper
npm error   replicad-threejs-helper@"^0.17.0" from the root project
npm error
npm error Fix the upstream dependency conflict, or retry
npm error this command with --force or --legacy-peer-deps
npm error to accept an incorrect (and potentially broken) dependency resolution.
npm error
npm error
npm error For a full report see:
npm error /home/user/.npm/_logs/2024-09-04T22_34_07_323Z-eresolve-report.txt
npm error A complete log of this run can be found in: /home/user/.npm/_logs/2024-09-04T22_34_07_323Z-debug-0.log
```

sgenoud, 2024-09-27:
I have updated the dependencies, it should work now!

mmiscool, 2024-09-29:
I can confirm that building the project works now.

closed 2024-09-29 by mmiscool