sgenoud/replicad #169: Studio: Failed to resolve import "replicad.d.ts?raw"

goswinr, 2024-10-03:
Hi,
I tried to build the studio following the [instructions ](https://github.com/sgenoud/replicad/tree/main/packages/studio). but have this Error. I ran `npm i` in the root too.

```
[plugin:vite:import-analysis] Failed to resolve import "../../node_modules/replicad/dist/replicad.d.ts?raw" from "src/workbench/EditorPane.jsx". Does the file exist?
C:/Dev/Sources/replicad/packages/studio/src/workbench/EditorPane.jsx:6:26
20 |  import { observer } from "mobx-react";
21 |  import Editor from "@monaco-editor/react";
22 |  import replicadTypes from "../../node_modules/replicad/dist/replicad.d.ts?raw";
   |                             ^
23 |  import Splitter, { GutterTheme, SplitDirection } from "@devbookhq/splitter";
24 |  import "./loadMonaco";
    at TransformPluginContext._formatError (file:///C:/Dev/Sources/replicad/packages/studio/node_modules/vite/dist/node/chunks/dep-CDnG8rE7.js:49193:41)
    at TransformPluginContext.error (file:///C:/Dev/Sources/replicad/packages/studio/node_modules/vite/dist/node/chunks/dep-CDnG8rE7.js:49188:16)
    at normalizeUrl (file:///C:/Dev/Sources/replicad/packages/studio/node_modules/vite/dist/node/chunks/dep-CDnG8rE7.js:63984:23)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async file:///C:/Dev/Sources/replicad/packages/studio/node_modules/vite/dist/node/chunks/dep-CDnG8rE7.js:64116:39
    at async Promise.all (index 7)
    at async TransformPluginContext.transform (file:///C:/Dev/Sources/replicad/packages/studio/node_modules/vite/dist/node/chunks/dep-CDnG8rE7.js:64043:7)
    at async PluginContainer.transform (file:///C:/Dev/Sources/replicad/packages/studio/node_modules/vite/dist/node/chunks/dep-CDnG8rE7.js:49034:18)
    at async loadAndTransform (file:///C:/Dev/Sources/replicad/packages/studio/node_modules/vite/dist/node/chunks/dep-CDnG8rE7.js:51867:27
Click outside, press Esc key, or fix the code to dismiss.
You can also disable this overlay by setting server.hmr.overlay to false in vite.config.js.
```

sgenoud, 2024-10-18:
I tried again and it works for me. I guess this is a Windows + vite issue, but I do not have a windows machine to debug it myself.

goswinr, 2024-10-22:
I solved it by changing this path: 

'"../../node_modules/replicad/dist/replicad.d.ts?raw"'

I will create a PR when back at my PC ...