sgenoud/replicad #173: Potential bug with clone and cut

alzatin, 2024-12-02:
Hello,
I'm experiencing some issues trying to return a shape that I've cloned and cut. There seems to be a misplaced wire on one of the shape faces. 
I've reproduced the issue on the workbench. 

[image]

Any ideas on what might be going on and if there's a way to fix it or if it's a bug?

Thanks!

BarbourSmith, 2024-12-18:
Any thoughts on this? I'm running into it fairly often :/

sgenoud, 2024-12-20:
Sorry, I missed it when you asked first!

Unfortunately, there isn't much I can do about it - this is an issue within the kernel that I use, as far as I can see.

One thing you could do is, instead of cloning the shape, [would be to create it twice](https://studio.replicad.xyz/workbench#code=UEsDBAoAAAAIAEA%252FlFnv3%252F%252FOwgAAAFYBAAAHAAAAY29kZS5qc1WOwQrCMAyG732K4KmFMVQQD2NefAAFL8rwUNrgirVK1qEie3ej01Uhh%252BTLn%252F%252BPOYcmwgMs6evSkfEIHZRAePHOaFsIYd6Kk3aBuVRQLlgOQgD0mxrdoY68m4wHRtq6tmE2G5C5excs0mCSEmUvV3lzxGjqVVh7HVCOtruRyvEWqbUo%252BxhVpOBIOjReR7Ts%252BbWXKh%252B4rOYZjLn2Kr3Rxk2tL%252Fh%252FwlQmu5eaizC2FKDiFn7EWT9%252FfFLaTk6mr8u96IonUEsBAhQACgAAAAgAQD%252BUWe%252Ff%252F87CAAAAVgEAAAcAAAAAAAAAAAAAAAAAAAAAAGNvZGUuanNQSwUGAAAAAAEAAQA1AAAA5wAAAAAA) (by creating a function).

Another way to handle that would be to [do the boolean operations in 2D and then extrude in 3D](https://studio.replicad.xyz/workbench#code=UEsDBAoAAAAIAMU9lFl%252F2lEhowAAABUBAAAHAAAAY29kZS5qc02PsQoCMQyG9z5FxhbkOAdxOM7FB1BwFIfQBls865G2KMi9u9VCK2T6%252Fi9%252FiH74EOENhvG5d6wnggVGYJonp9EMQuifcUfnM5cKxl3WQQiAklhyVxtztu4rYzQuhcw2FelSPv5dkkVTrSsy%252BjBhJJPFstFVJs%252FbFfQX1SpTPFmcqbmZyNbxNfMwxcS%252B2l24UdT24I8TepKqo1fkZEiWR5RYhg9QSwECFAAKAAAACADFPZRZf9pRIaMAAAAVAQAABwAAAAAAAAAAAAAAAAAAAAAAY29kZS5qc1BLBQYAAAAAAQABADUAAADIAAAAAAA%253D).

BarbourSmith, 2024-12-20:
Gotcha! That makes senses, but its a bummer.

Do you know what exactly is going on? Like when does it happen? Is the issue caused by cloning?

We have sort of limited control over how it comes up because we're mostly running into it in situations like this (although in this case I couldn't make it happen):

[image]

So we are somewhat constrained in terms of how we can tackle it

sgenoud, 2025-01-10:
One thing I could see (not the most efficient approach) would be to handle cloning differently (run twice your pipeline).

BarbourSmith, 2025-01-10:
That's a very interesting idea.

How much of the pipeline would need to be re-run? Would we need to go back the whole distance or just a few steps?

What is it about clone here which is creating the issue? Is there some identifier which is causing the bug, or is it the geometry itself being identical which is the problem?

sgenoud, 2025-01-11:
This is all part of the kernel's black arts that I don't know enough about.

The bug looks like there is an issue with identity (i.e. things that are kept the same but they should not), so that's why I would go with regenerating thing (as long as it is not too expensive).

You'll need to experiment with it I think unfortunately!

BarbourSmith, 2025-01-11:
I'm glad that you think of the kernel as magic too, because that's where I'm at 😂

I was thinking the same thing about something getting duplicated in clone. It seems unlikely that the issue is truly a geometry thing because if it's generated separately the issue doesn't show up, so I bet that you are right that there is a work around other than using clone which will fix it. 

We will have to play around with it an explore :)