sgenoud/replicad #174: Incorrect Projections for Shapes Cut from a Curved Mask

georgslazdans, 2024-12-19:
Hello,
I have found issue with projections for a shape that is cut from a curved mask. The edges are not symmetrical and are pointy at the top. After importing the shape as a STEP file in FreeCad I got the expected results.

## FreeCad projection
[image]

## Replicad projection
[image]

## Code to reproduce the issue
```javascript
const { draw, drawCircle, drawRectangle, drawProjection } = replicad;

const defaultParams = {
  materialWidth: 12,
  sideBoardCount: 2,
  topPadding: 50,
};

const roundingMask = ({ bottomWidth, bottomHeight, topPadding }) => {
  const radius = bottomWidth / 2;
  return draw()
    .vLine(bottomHeight + topPadding)
    .bezierCurveTo(
      [radius, 0],
      [
        [bottomWidth / 2, bottomHeight],
        [bottomWidth / 2, bottomHeight / 2],
      ]
    )
    .close()
    .sketchOnPlane("XZ")
    .revolve()
    .translateX(bottomWidth / 2);
};

const cutSideBoardsFromMask = (
  mask,
  { legMaterialWidth, sideMaterialWidth, sideBoardCount }
) => {
  const size = 20000;
  const sideBoard = drawRectangle(sideMaterialWidth, size)
    .sketchOnPlane("YZ")
    .extrude(size)
    .translateZ(size / 2)
    .translateX(-size / 2);

  const middleOffset = legMaterialWidth / 2 + sideMaterialWidth / 2;
  const resultingBoards = [];
  for (let i = 0; i < sideBoardCount; i++) {
    const offset = middleOffset + sideMaterialWidth * i;
    const board = sideBoard.clone().translateY(offset);
    resultingBoards.push(mask.intersect(board));
  }

  for (let i = 0; i < sideBoardCount; i++) {
    const offset = -middleOffset - sideMaterialWidth * i;
    const board = sideBoard.clone().translateY(offset);
    resultingBoards.push(mask.intersect(board));
  }

  sideBoard.delete();

  return resultingBoards;
};

const main = async (r, { materialWidth, sideBoardCount, topPadding }) => {
  const legConfig = {
    materialWidth: 12,
    bottomHeight: 120
  }
  const bottomWidth =
    legConfig.materialWidth + materialWidth * (sideBoardCount * 2);

  const mask = roundingMask({
    bottomWidth,
    bottomHeight: legConfig.bottomHeight,
    topPadding,
  });

  const shape =  cutSideBoardsFromMask(mask, {
    legMaterialWidth: legConfig.materialWidth,
    sideMaterialWidth: materialWidth,
    sideBoardCount: sideBoardCount,
  })[1]
  .translateX(-bottomWidth/2).translateY(-materialWidth);

  // return shape;
  return drawProjection(shape, "front").visible;
};
```

sgenoud, 2024-12-20:
Thanks!

I have found the issue (which was with the approximation done to convert to SVG). I also exposed a method if you want a better / worse approximation:

```js
  return drawProjection(shape, "front").visible.approximate("svg", {
    tolerance: 1
  })
}
```

georgslazdans, 2024-12-20:
Wow, that was a fast fix. I updated the Replicad dependency in my project and the exported PDF template from the projection looks nice with the default values. 

After playing a bit with the tolerance value in the studio, it also seems to be good with high values. Very nice!

closed 2024-12-20 by georgslazdans