sgenoud/replicad #175: Issues with Cutting Small Rounded Lofted Shapes

georgslazdans, 2024-12-23:
# Description
The issue is that under certain conditions the cut operation doesn't do anything or returns just the mask object. When the cut succeeds the narrow part gets clipped. I observed that this happens when either the material is the same width as the mask or the mask itself was translated, but only the width axis and not the the length axis.
[image]
[image]

I am trying to achieve a seam between a circle base and a extruded rectangle, that will have a fillet on the edge (e.g. rounded plywood board). Creating the mask shape using FreeCad works as expected, albeit it is defined by the length of the sides and distance to the circle center.
[image]
[image]

# Example code

```javascript
const { draw, drawCircle, drawRectangle } = replicad;

const defaultParams = {
  seamHeight: 10,
};

const toDegrees = (radians) => {
  return (radians * 180) / Math.PI;
};

const cornerCutOffMask = ({
  seamHeight,
  materialHalfWidth,
  fillet,
  bottomRadius,
}) => {
  const filletOffset = Math.sqrt(fillet * fillet + fillet * fillet) - fillet;
  const topSketch = draw()
    .vLine(fillet)
    .threePointsArcTo([fillet, 0], [filletOffset, filletOffset])
    .close()
    .sketchOnPlane("XY", seamHeight);

  const bottomAngle = Math.atan(materialHalfWidth / bottomRadius);
  const offset = bottomRadius - Math.cos(bottomAngle) * bottomRadius;
  const bulge = toDegrees(bottomAngle) / 180;
  const bottomSketch = draw()
    .vLine(materialHalfWidth)
    .bulgeArcTo([offset, 0], bulge)
    .close()
    .sketchOnPlane("XY", 0);

  return bottomSketch.loftWith(topSketch);
};

const main = async (r, { seamHeight }) => {
  const legConfig = {
    materialWidth: 12,
    fillet: 1,
  };
  const legRoundingConfig = {
    materialWidth: 12,
    sideBoardCount: 2,
  };
  const bottomWidth =
    legConfig.materialWidth +
    legRoundingConfig.materialWidth * (legRoundingConfig.sideBoardCount * 2);

  const fillet = legConfig.fillet;
  const materialHalfWidth = legConfig.materialWidth / 2;
  const bottomRadius = bottomWidth / 2;

  const cornerMask = cornerCutOffMask({
    seamHeight,
    materialHalfWidth,
    fillet,
    bottomRadius,
  });

  // This works
  const legExample = drawRectangle(200, legConfig.materialWidth)
    .sketchOnPlane()
    .extrude(100)
    .translateX(100)
    .translateY(materialHalfWidth);
  return legExample.cut(cornerMask);

  // This works, half width with small marging
  const legWorking2 = drawRectangle(200, legConfig.materialWidth / 2 + 0.1)
    .sketchOnPlane()
    .extrude(100)
    .translateX(100)
    .translateY(legConfig.materialWidth / 4 + 0.05);
  // return legWorking2.cut(cornerMask);

  // This doesn't work, half width
  const legNotWorking = drawRectangle(200, legConfig.materialWidth / 2)
    .sketchOnPlane()
    .extrude(100)
    .translateX(100)
    .translateY(legConfig.materialWidth / 4);
  return legNotWorking.cut(cornerMask);
};
```
## Mask code
I am creating the masks for each corner
```javascript
const leftSideMask = ({
  seamHeight,
  fillet,
  filletOffset,
  materialHalfWidth,
  offset,
  bulge,
}) => {
  const bottomSketch = draw()
    .hLine(offset)
    .bulgeArcTo([0, -materialHalfWidth], bulge)
    .close()
    .sketchOnPlane("XY", 0);

  const topSketch = draw()
    .hLine(fillet)
    .threePointsArcTo([0, -fillet], [filletOffset, -filletOffset])
    .close()
    .sketchOnPlane("XY", seamHeight);

  return bottomSketch.loftWith(topSketch).translateY(materialHalfWidth);
};

const rightSideMask = ({
  seamHeight,
  fillet,
  filletOffset,
  materialHalfWidth,
  offset,
  bulge,
}) => {
  const bottomSketch = draw()
    .vLine(materialHalfWidth)
    .bulgeArcTo([offset, 0], bulge)
    .close()
    .sketchOnPlane("XY", 0);

  const topSketch = draw()
    .vLine(fillet)
    .threePointsArcTo([fillet, 0], [filletOffset, filletOffset])
    .close()
    .sketchOnPlane("XY", seamHeight);

  return bottomSketch.loftWith(topSketch).translateY(-materialHalfWidth);
};

const backRightSideMask = ({
  seamHeight,
  fillet,
  filletOffset,
  materialHalfWidth,
  offset,
  bulge,
}) => {
  const bottomSketch = draw()
    .vLine(materialHalfWidth)
    .bulgeArcTo([-offset, 0], -bulge)
    .close()
    .sketchOnPlane("XY", 0);

  const topSketch = draw()
    .vLine(fillet)
    .threePointsArcTo([-fillet, 0], [-filletOffset, filletOffset])
    .close()
    .sketchOnPlane("XY", seamHeight);

  return bottomSketch.loftWith(topSketch).translateY(-materialHalfWidth);
};

const backLeftSideMask = ({
  seamHeight,
  fillet,
  filletOffset,
  materialHalfWidth,
  offset,
  bulge,
}) => {
  const bottomSketch = draw()
    .vLine(-materialHalfWidth)
    .bulgeArcTo([-offset, 0], bulge)
    .close()
    .sketchOnPlane("XY", 0);

    const topSketch = draw()
    .hLine(fillet)
    .threePointsArcTo([0, fillet], [filletOffset, filletOffset])
    .close()
    .rotate(180)
    .sketchOnPlane("XY", seamHeight);

  return bottomSketch.loftWith(topSketch).translateY(materialHalfWidth);
};
```

sgenoud, 2025-01-10:
Sorry, this is also one of the issue I am going to blame on the kernel. In theory it should have the same issue with FreeCAD, but it is possible that they have a better version of the kernel (mine has not been updated for a while unfortunately).

That said, there is a dirty hack for this kind of situation. If you do more a little bit your parts it sometimes helps the kernel to do its job. For instance, you can do:

```js
  const legNotWorking = drawRectangle(200, legConfig.materialWidth / 2)
    .sketchOnPlane()
    .extrude(100)
    .translateX(100 + 1e-12)
    .translateY(legConfig.materialWidth / 4);
```

and it fixes it. I told you it was a hack...

georgslazdans, 2025-01-11:
Need a bit higher offset and the smallest details are still missing, but should be good enough for woodworking needs. Thank you!