sgenoud/replicad #176: Show previous model on error

North101, 2025-01-09:
When an error happens, a placeholder is shown. This can be jarring or make it difficult to see the change made if it tries to render while in the middle of typing.

I think it'd be better if it showed the previous model with the previous camera position with some indication that an error happened (e.g. a red border) instead. 

Thanks for the great project!

sgenoud, 2025-01-11:
I like the blob! I see your point though, but I like to know when my code breaks something. So I can be sure that if I see something working, that it corresponds to a working state.

I have faced an issue like yours (I think), when I want to keep having a look at some drawing before extruding it. Perhaps having a way to have different views that are rendered individually might be a better solution to your problem.

In any case I will keep thinking about it (while probably keeping the blob here)

North101, 2025-01-11:
I also like the blob but when making small changes having the model disappear and the camera reset makes it hard to see what change has been made.

Jamesgt, 2025-01-17:
Maybe a red border around the preview or a popup can show that the code does not compile. But it does not make too much sense that when I start typing `totalHeight` then I get 10 rerender of the blob and finally the result.