sgenoud/replicad #182: How to create ellipsoid?

ikeji, 2025-01-18:
Sorry for random question.
How to create ellipsoid?

raydeleu, 2025-01-18:
This question probably belongs under "discussions"? My first take on this would be to revolve an ellipse. If that is sufficient depends on the type of ellipsoid that you need. The scale function that can be used on shapes, such as a sphere, currently only seems to support uniform scaling.

ikeji, 2025-01-20:
> This question probably belongs under "discussions"? 
Oops sorry, I created #183 

> My first take on this would be to revolve an ellipse. If that is sufficient depends on the type of ellipsoid that you need. 

I want to have ellipsoid with 3 radius.
I think we can't create it with revolve.

> The scale function that can be used on shapes, such as a sphere, currently only seems to support uniform scaling.

Yea, scale with 1 axis only would be more flexible way, but the scale function doesn't have the option.

closed 2025-01-20 by ikeji