sgenoud/replicad #185: Missing defaultParams leva processing on first run

Jamesgt, 2025-02-04:
Check the following code's output, that tries to use the leva lib's folders and display names feature.

```javascript
const defaultParams = {
  a: 1,
  "b.c": 2,
  "b.d": { label: "d", value: 3 },
};

const main = (R, p) => {
  console.log(p);
  return [{ shape: R.makeBaseBox(10, 10, 10) }];
};
```

On the first, vanilla run it is:

```json
{
  "a": 1,
  "b.c": 2,
  "b.d": {
    "label": "d",
    "value": 3
  }
}
```

After you press the Apply params button, the output changes to (the expected) format:

```json
{
  "a": 1,
  "b.c": 2,
  "b.d": 3
}
```

sgenoud, 2025-02-07:
My use of leva is a bit more of an implementation detail for now. There are a bunch of rough edges to make the data models fit together I am not very happy with (and might want to change later on).

If you still want to rely on more advanced features of it, I have seen [members of the community hack around my system](https://www.printables.com/model/702716-parametric-box-with-lid).

I can't promise it will keep working, but for now this should be your workaround.

Would you mind if I convert this to a discussion - this feels more appropriate if people want to discuss this further.

Jamesgt, 2025-02-10:
Of course, let it be a discussion.