sgenoud/replicad #186: Add typescript support

Jamesgt, 2025-02-04:
I prefer to use TS in my codes, so there is a build step now, but this way I cannot easily share my code.

Maybe adding a setting/drop-down/auto-detect feature and using the ts.transpile method would be a trivial addition and might help getting more users.

mordv, 2025-04-05:
It seems trivial - one needs to add 
```js
import { transpile } from 'typescript';
...
const t = transpile(code);
```
In `vm.js`. And add "typescript" to "dependencies" in package.json. Works for me. No drop-down/autodetection needed.

sgenoud, 2025-04-11:
I have tested your method - and it works for the transpilation part. 

It adds about 3MB to download. On one side it is a third of the opencascade - but on the other it is still quite big.

Note that in order to have a nice behaviour there is a need to specify if one works within TS or JS for the type checking within the monaco editor.

I probably should do something that only loads typescript transpilation when needed (but how to detect this I am not sure). 

I have put my current work in a branch!

Jamesgt, 2025-04-11:
Instead of detection I would still recommend a UI control.
With a quick research i have not found any good solution.
Maybe you can use the file extension if the external file feature is used... btw is it allowed to browse other than .js files?

North101, 2025-05-05:
This is great! Its showing type errors though with `cut` and `fuse`.

```typescript
drawRectangle(width, height, corner)
      .sketchOnPlane('XY')
      .extrude(depth)
      .cut(drawRectangle(width - (thickness * 2), height - (thickness * 2))
        .sketchOnPlane('XY')
        .extrude(depth - thickness)
        .translateZ(thickness)
      )
```

[image]
[image]

This is because `sketchOnPlane` returns `SketchInterface | Sketches` and each has different return types for `extrude`. Is there a way to fix this without casting?

sgenoud, 2025-05-09:
This is something I have been thinking about - I think I have an idea on how to handle it, but I am not really happy about it.

The core of the issue is multiple:

* a drawing can be a set of disjoint closed shapes (typically if you cut a shape with another you can easily get into this case). This is the reason that sketchOnPlane does not return just a `SketchInterface`.
* operation that extrude or do some boolean operation can also return some unexpected shape (if you extrude a sketch that is not closed you will have a shell / face that behaves differently than a simple solid).

That said, in many cases, while the code cannot guess the precise type of the result, the developper knows what to expect. In that case casting makes sense (i.e. the user knows more than the type system).

The other solution would be to offer a method that throws if the output is not a single sketch. I might need to break the compatibility of the API (i.e. one would expect the default to be a single sketch and allow for complex sketch with a configuration).

johncomposed, 2025-10-17:
I've been experimenting a bit with this and a quick good-enough solution I've found is to add a little guard method `.isShape3D()`. I'm doing it via monkey-patching in my app, but this could be integrated upstream fairly easily.   

```ts
/* in replicad.d.ts */
import {
  Solid, Shell, Compound, CompSolid, Face, Wire, Edge, Vertex,
  _3DShape, Shape, TopoDS_Solid, TopoDS_Shell, TopoDS_Compound,
  TopoDS_CompSolid, TopoDS_Face, TopoDS_Wire, TopoDS_Edge, TopoDS_Vertex,
  WrappingObj
} from "replicad";

declare module "replicad" {

  // When monkey-patching, we have to re-declare the inheritance to preserve the link to the parent.
  interface Shape<Type extends TopoDS_Shape> extends WrappingObj<Type> { isShape3D(): this; }

  // --- 3D Shapes ---
  interface Solid extends _3DShape<TopoDS_Solid> { isShape3D(): this; }
  interface Shell extends _3DShape<TopoDS_Shell> { isShape3D(): this; }
  interface Compound extends _3DShape<TopoDS_Compound> { isShape3D(): this; }
  interface CompSolid extends _3DShape<TopoDS_CompSolid> { isShape3D(): this  ; }

  // --- Non-3D Shapes ---
  interface Face extends Shape<TopoDS_Face> { isShape3D(): never; }
  interface Wire extends Shape<TopoDS_Wire> { isShape3D(): never; }
  interface Edge extends Shape<TopoDS_Edge> { isShape3D(): never; }
  interface Vertex extends Shape<TopoDS_Vertex> { isShape3D(): never; }
}

/* in replicad-patch.ts, imported at the start of my app */
import { Shape, isShape3D, Shape3D } from "replicad";

(Shape as any).prototype.isShape3D = function (): Shape3D {
  if (isShape3D(this)) return this;
  throw new Error(
    `Type Guard Error: Expected a 3D Shape, but got a ${this.constructor.name}.`
  );
};
```

When using, I just write something like:
```ts
export function drawBox(thickness: number) {
  return drawRoundedRectangle(30, 50)
    .sketchOnPlane()
    .extrude(20)
    .isShape3D()
    .shell(thickness, (f) => f.inPlane("XY", 20));
}

function drawCutRectangle(width, height, corner, depth, thickness) {
  return drawRectangle(width, height, corner)
    .sketchOnPlane('XY')
    .extrude(depth)
    .isShape3D()
    .cut(drawRectangle(width - (thickness * 2), height - (thickness * 2))
      .sketchOnPlane('XY')
      .extrude(depth - thickness)
      .isShape3D()
      .translateZ(thickness)
    )
}
```