sgenoud/replicad #192: Workbench and share failing to load in Firefox

Billiam, 2025-03-11:
When visiting the share page in firefox (135.0.1) for the default vase object, both workbench link:

https://studio.replicad.xyz/workbench#code=UEsDBAoAAAAIAN0Ga1otIUITHAEAAIUCAAAHAAAAY29kZS5qc6WSzWrDMBCE736KJSc5GGOnlEBMeuyp0EAD%252FQk5qPY6ElWkIG3TQ%252FC7d%252B3YbnszVMcZdr5lVlHpbCC4QOXlFzSwBo8no0tZFVF09Y5SW9ZFDOs7uEQAV%252FldBnzWFSn2Flkx6gr1QRGLecbqKJ%252B8q7VB1luSiNkASNWDtijGqF4NR%252BdIPfEeFrdO7H5Qc8jT22RgzCFLF%252FukW6p9aKuttAe0tIJdlkC%252BTzqnmZCbpcs%252FucsJuQCBpKd7WZLzK7iZShtA%252F0YM0dmY2BulcQFF3PXvkT69HQ7Q7%252FaBVKpHuzGS%252B5%252B9vM36QY9nZ8443CcoNEZw46Luzl%252BnfE3iDxE2TlvqyL%252Fg%252FRRzDJLI20oFdoOY6hH2OhsmYt6wKaJvUEsBAhQACgAAAAgA3QZrWi0hQhMcAQAAhQIAAAcAAAAAAAAAAAAAAAAAAAAAAGNvZGUuanNQSwUGAAAAAAEAAQA1AAAAQQEAAAAA

and the share page

https://studio.replicad.xyz/share/code/?ortho-camera=true#code=UEsDBAoAAAAIAN0Ga1otIUITHAEAAIUCAAAHAAAAY29kZS5qc6WSzWrDMBCE736KJSc5GGOnlEBMeuyp0EAD%252FQk5qPY6ElWkIG3TQ%252FC7d%252B3YbnszVMcZdr5lVlHpbCC4QOXlFzSwBo8no0tZFVF09Y5SW9ZFDOs7uEQAV%252FldBnzWFSn2Flkx6gr1QRGLecbqKJ%252B8q7VB1luSiNkASNWDtijGqF4NR%252BdIPfEeFrdO7H5Qc8jT22RgzCFLF%252FukW6p9aKuttAe0tIJdlkC%252BTzqnmZCbpcs%252FucsJuQCBpKd7WZLzK7iZShtA%252F0YM0dmY2BulcQFF3PXvkT69HQ7Q7%252FaBVKpHuzGS%252B5%252B9vM36QY9nZ8443CcoNEZw46Luzl%252BnfE3iDxE2TlvqyL%252Fg%252FRRzDJLI20oFdoOY6hH2OhsmYt6wKaJvUEsBAhQACgAAAAgA3QZrWi0hQhMcAQAAhQIAAAcAAAAAAAAAAAAAAAAAAAAAAGNvZGUuanNQSwUGAAAAAAEAAQA1AAAAQQEAAAAA

Fail to load and show a white screen, with this error:

```
Uncaught Error: Minified React error #301; visit https://reactjs.org/docs/error-decoder.html?invariant=301 for the full message or use the non-minified dev environment for full errors and additional helpful warnings.
```

[image]

The workbench itself loads successfully when not using the code parameter, displaying the default vase object.

Billiam, 2025-03-11:
This began with this change: https://github.com/sgenoud/replicad/commit/418c543837f7835f8c2ce32f57a612897ee23e74#diff-014e19a9c3a73053c3d91accaca6c87aa3bc70718c0982510e8b86c19e842087R57-R60

I'm not sure what the correct logic for this is, but if the `shapeURL` is empty (because you're using the `/share/code` URL), this block returns early. Commenting out this check instead allows the `code` param path to run correctly

sgenoud, 2025-03-12:
Thanks! I have fixed it, it should work fine now!

closed 2025-03-12 by Billiam