sgenoud/replicad #198: Request: Support multi-object STLs

Billiam, 2025-03-31:
Currently, the STL export options return a zip file when a project has multiple shapes (ex: when `main` returns an array).

For some 3d print projects, it can be helpful to have a single STL file with relevant objects pre-aligned.

My typical use case for this is multimaterial text, where an object (like a box lid) has the text engraved into it, alongside separate objects for the text itself.

When printing, a user would import that single STL, split the object in their slicer application, and assign the text objects a different filament/color before printing (using either an AMS/MMU unit, or by manually swapping filament during printing).

This is not a huge issue, since a) the user _can_ use assembly step files if they know to or b) import multiple STLs in a way that maintains the alignment as well (some slicers will automatically center imported objects), but a single STL file for this is more common/expected, and less error-prone.

Billiam, 2025-04-04:
Realized this could be accomplished using the undocumented `makeCompound` method.

closed 2025-04-04 by Billiam

sgenoud, 2025-04-04:
Also with STEP files you can go for a STEP assembly that returns a compound (and also adds the colours and name into the file).