sgenoud/replicad #201: [request] Add support for asymmetrical chamfers

Billiam, 2025-04-07:
I think it would be be useful to be able to create non-45-degree chamfers via `BRepFilletAPI_MakeChamfer.add_3` or `.addDA`, which either accept two distances for the chamfer, or a single distance and an angle.

I'm not sure what the interface would be like for this, since the methods require one of the connecting faces as an argument to orient the chamfer.

I guess it could accept a faceFinder, but maybe there's a simpler way?

sgenoud, 2025-04-11:
I had structured my code in prevision of doing something in that spirit, so I just implemented it.

[image]

You can play with this [on the workbench](https://studio.replicad.xyz/workbench#code=UEsDBAoAAAAIACdhi1repY40qgEAAEsEAAAHAAAAY29kZS5qc7VT30vDMBB%252B719x9CmFOuaPgWzow9Q9C%252Fqgjj7E9OqCaVqSDAey%252F91bknYdU9AHCyXkfnz3fXcX0Wjr4BNq%252Fo5zbnHebHIQTf0qNS6kLtEspHJobA535Vs0wRauwGCrpODlLElw0zbGQYkVXysH1VoLJxtNoFKzDD4TAIUOXgmfEgel2Pk4hwn9p%252BNsRlHCs7GybtUucpcwEqrRyLKRWPG6QsMoDDzk7iuldVwLtFNYTnI4K%252FLosKhQOCwXXOAUWJXB1TVUI6nvFSe49PklpaqTLMRvw8HQh%252BE%252B7Ok59eTITQR7hsvKNyWHcN4o5Logwt81ji0PGBteyjXR7e6HGi6p2l7EX2X0QiASm4LGj8HgWNYjH2PsXbpxrO9Fyw1XCtVjE9sxzg679jtl5zTo%252FxR2PLMByWI4PBoSrdfmp%252F0KBfw6DqbLhghk3%252B1zAPBVOpRwCX62vCAeRf7NWnXqyHIrDfrXQtY0C1UMurXREBcH7Iq31KTwLHLQvKZb%252BhBeSaSd9tPowqPMkTNcW8UdPrELakqXfhO78GN%252BEDFMPxnmL7w7phWzZJt8AVBLAQIUAAoAAAAIACdhi1repY40qgEAAEsEAAAHAAAAAAAAAAAAAAAAAAAAAABjb2RlLmpzUEsFBgAAAAABAAEANQAAAM8BAAAAAA%253D%253D)

You can now configure the chamfers with custom distances like that:

```js
const radius = {
    distances: [2, 6],
    selectedFace: f => f.inPlane("XY")
}
```

This is a power tool API - there are many ways to make your chamfers break that way, I'm sure, but I don't check for anything (if you have some common cases to check, I would be happy to see to add them). The definition is [here](https://replicad.xyz/docs/api/#chamferradius), you can see there is also a definition for `{distance: 2, angle: 45, selectedFace: fn}`.

For complex fillets with multiple different radiuses you can still use the `combineFinderFilters ` helper to build a function that returns different radiuses for different edges, as [explained in the documentation](https://replicad.xyz/docs/tutorial-overview/modifications#multiple-radii)

I also added something similar for fillets, where you can now pass an array of numbers to have a fillet with varying radius.

Billiam, 2025-04-11:
Incredible! Thank you!

closed 2025-04-14 by Billiam