sgenoud/replicad #219: Bug: Make enough horizontal space for defaultParams labels

danfuzz, 2025-08-05:
It looks like when one has long `defaultParams` labels (or just names-as-labels), they can get ellipsized, as in this example:

```js
const defaultParams = {
  topLineXAxis:     { value: false, label: 'top: x axis channel' },
  topLineYAxis:     { value: false, label: 'top: y axis channel' },
  topCenterBore:    { value: false, label: 'top: center bore' },
  middleBoreXAxis:  { value: false, label: 'middle: x axis bore' },
  middleBoreYAxis:  { value: false, label: 'middle: y axis bore' },
  bottomLineXAxis:  { value: false, label: 'bottom: x axis channel' },
  bottomLineYAxis:  { value: false, label: 'bottom: y axis channel' },
  bottomCenterBore: { value: false, label: 'bottom: center bore' }
};
```

[image]

It'd be great if space were made for the labels automatically, or if the end user had a way to stretch the window, or if the API user could control the width.

danfuzz, 2025-08-06:
Based on a little experimentation, I think the specification `max-width: 300px` on line 9 of [FloatingInfo.jsx](https://github.com/sgenoud/replicad/blob/main/packages/studio/src/components/FloatingInfo.jsx#L9) can simply be removed, and the params popup will get as wide as needed to fit the parameter labels.

sgenoud, 2025-08-15:
Thanks, fixed !

[image]

closed 2025-08-15 by sgenoud