sgenoud/replicad #221: Bug when two intersecting cylinders are used to cut

danfuzz, 2025-08-05:
[This is presumably a bug in the underlying OpenCASCADE library, but I don't know how to translate my findings here to that level.]

I ran into a modeling issue where two intersecting cylindrical cuts into a solid caused the surface created from the cutouts to be incorrect. In particular, it looks like a little sliver of the surface is missing, such that one can see through the solid part to the "outside," though when looking from the outside one doesn't see in. Here's a picture of what I mean:

[image]

Here's the code I used to create this shape. Notably, reversing the order of cuts makes the problem go away.

```js
const {
  drawCircle,
  drawRectangle
} = replicad;

const SIZE_MM          = 30;
const HOLE_RADIUS_MM   = 10;
const HALF_SIZE_MM     = SIZE_MM / 2;

function main() {
  const base = drawRectangle(SIZE_MM, SIZE_MM)
    .sketchOnPlane('XY')
    .extrude(SIZE_MM)
    .translateZ(-HALF_SIZE_MM);
  const yTube = drawCircle(HOLE_RADIUS_MM)
    .sketchOnPlane('XZ')
    .extrude(SIZE_MM)
    .translateY(HALF_SIZE_MM);
  const zTube = drawCircle(HOLE_RADIUS_MM)
    .sketchOnPlane('XY')
    .extrude(HALF_SIZE_MM);

  // Uncomment this to see a version that works.
  //return base.cut(yTube).cut(zTube);

  return base.cut(zTube).cut(yTube);
}
```

This isn't just a rendering glitch: I exported the model as an STL and pulled it into Cura for slicing, and the resulting sliced model showed missing material in the problem area.

danfuzz, 2025-08-07:
In case it is of use, I ended up working around the `fuse()` version of this issue with the following function, which tries `fuse()`ing in both directions, ignores errors that only happen in one of the directions, and takes the result with the most edges if both directions produce a result (because presumably the result with more edges is the correct one).

```js
function fuseAll(...shapes) {
  switch (shapes.length) {
    case 0: {
      return null;
    }
    case 1: {
      return shapes[0];
    }
    case 2: {
      // Do a `fuse()` in both directions, in order to work around a couple bugs
      // (or one bug which shows up in two ways) in Open Cascade: Sometimes a
      // `fuse()` in one direction totally fails (throws an error) while the
      // other way works. And sometimes both "work" except one way produces
      // a result with more edges, and the less-edged result actually
      // incorrectly omits at least one edge.

      let way1 = null;
      let way2 = null;
      let error1 = null;
      let error2 = null;

      try {
        way1 = shapes[0].fuse(shapes[1]);
      } catch (e) {
        error1 = e;
      }

      try {
        way2 = shapes[1].fuse(shapes[0]);
      } catch (e) {
        error2 = e;
      }

      if (way1 && way2) {
        const edges1 = way1.edges.length;
        const edges2 = way2.edges.length;
        return (edges1 >= edges2) ? way1 : way2;
      } else if (way1) {
        return way1;
      } else if (way2) {
        return way2;
      } else {
        // Both ways threw an exception. Pick one.
        throw error1;
      }
    }
    default: {
      // Three or more shapes: Using recursive calls, make pairwise fuses, and
      // then combine all the pairs plus any oddball leftover.
      const newShapes = [];
      let   pending = null;

      for (const s of shapes) {
        if (pending) {
          newShapes.push(fuseAll(pending, s));
          pending = null;
        } else {
          pending = s;
        }
      }

      if (pending) {
        newShapes.push(pending);
      }

      return fuseAll(...newShapes);
    }
  }
}
```

sgenoud, 2025-08-15:
Yes, you are right, this is an issue with the kernel. This tends to happen when you have some fusing when you do some booleans on things that have an exact same dimension (here your diameter).

A quick fix is to add a fuzz number, for instance:

```js
const {
  drawCircle,
  drawRectangle
} = replicad;

const SIZE_MM          = 30;
const HOLE_RADIUS_MM   = 10;
const HALF_SIZE_MM     = SIZE_MM / 2;

function main() {
  const base = drawRectangle(SIZE_MM, SIZE_MM)
    .sketchOnPlane('XY')
    .extrude(SIZE_MM)
    .translateZ(-HALF_SIZE_MM);
  const yTube = drawCircle(HOLE_RADIUS_MM)
    .sketchOnPlane('XZ')
    .extrude(SIZE_MM)
    .translateY(HALF_SIZE_MM);
  const zTube = drawCircle(HOLE_RADIUS_MM + 1e-7) // LOOK here there is a change
    .sketchOnPlane('XY')
    .extrude(HALF_SIZE_MM);

  // Uncomment this to see a version that works.
  //return base.cut(yTube).cut(zTube);

  return base.cut(zTube).cut(yTube);
}
```

it just fixes it. Welcome to the world of open cascade 😉